publiclab/webjack #17: adapt SoftModem lib for ATTiny chips

jywarren, 2016-07-23:
https://publiclab.org/notes/rmeister/07-18-2016/webjack-testers-needed#c15006

 looking for ATTiny softmodem, i find:

https://github.com/l1q1d/SoftModem-for-attiny85
http://fab.cba.mit.edu/classes/863.12/people/Adam.Marblestone/AHM_week11.html

Maybe we can port in the mods which made it work on the Tiny architecture: https://github.com/l1q1d/SoftModem-for-attiny85/commit/b6f08ca9ade1263b788dcde94fb0f4aefa65431c

Also happy to move this to your SoftModem fork instead of having it here

rmeister, 2016-07-23:
Great, thanks for the attiny85 hint! Didn't had that on my radar. Will open a new issue at SoftModem and see what I can do. Unfortunately I don't have an ATTiny, so I can't test it.

jywarren, 2016-07-23:
I'm happy to test, and anyone who has a tinyduino or a digispark can help;
maybe we can ask in one of their forums too.

Also, it might be needed to do the same for Leonardo based arduinos, so
perhaps we can keep that in mind both as another issue to open and in terms
of the code we write and how it's structured (for different chips).

rmeister, 2016-07-23:
Just pushed the new branch 'attiny-support'. You can give it a try, it compiles. 

Remeber to:

- add the ATTiny to the Arduino IDE: http://highlowtech.org/?p=1695
- change the frequencies in `SoftModem.h`
- use `SoftwareSerial` instead of `Serial` on the ATTiny

Also I'm not sure if the pins for RX(0) and TX(1) are correct.

This is the sketch I was able to compile:
https://gist.github.com/rmeister/3a2ff5a8ac6ad6eadd502d9e93ff9b19

jywarren, 2016-07-25:
Hmm, i get a bunch of errors like: 

`/Users/warren/Documents/Arduino/libraries/SoftModem-attiny-support/SoftModem.cpp:113:14: error: 'TCNT2' was not declared in this scope
  _lastTCNT = TCNT2;`

I can test again tomorrow morning, thanks for doing this!

jywarren, 2016-07-25:
https://gist.github.com/jywarren/dd68cda0b0dcc0462ef57023c1f3234c

rmeister, 2016-07-25:
Then probably this did not work:
```Arduino
#if defined(__AVR_ATtiny24__) || defined(__AVR_ATtiny44__) || defined(__AVR_ATtiny84__) || defined(__AVR_ATtiny25__) || defined(__AVR_ATtiny45__) || defined(__AVR_ATtiny85__)
#define ATTINY 1
#endif
```
Can you verify that `__AVR_ATtiny85__` exists,  maybe by using it in a sketch?

jywarren, 2016-07-25:
yes but see my latest response to your research note; i think perhaps there's something more basic wrong...

jywarren, 2016-07-25:
This compiled:

````
void setup() {
  // put your setup code here, to run once:

  int i = __AVR_ATtiny85__;

}

void loop() {
  // put your main code here, to run repeatedly:

}
````

Let me double check that I'm compiling for the right board...

jywarren, 2016-07-25:
Taking a closer look at the software serial implementation for the Nanite, I've gotten this to run: https://codebender.cc/sketch:279140#Nanite841%20SoftSerial.ino so i'll adapt that.

jywarren, 2016-07-25:
With that Nanite841 sketch, I got to this error: 

````
In file included from /Users/warren/Documents/Arduino/sketch_jul25a/sketch_jul25a.ino:2:0:
/Users/warren/Documents/Arduino/libraries/SoftModem-attiny-support/SoftModem.h:62:20: error: conflicting return type specified for 'virtual size_t SoftModem::write(const uint8_t*, size_t)'
     virtual size_t write(const uint8_t *buffer, size_t size);

                    ^
In file included from /Users/warren/Library/Arduino15/packages/digistump/hardware/avr/1.6.7/cores/tiny/Stream.h:24:0,
                 from /Users/warren/Library/Arduino15/packages/digistump/hardware/avr/1.6.7/cores/tiny/TinyDebugSerial.h:31,
                 from /Users/warren/Library/Arduino15/packages/digistump/hardware/avr/1.6.7/cores/tiny/WProgram.h:18,
                 from /Users/warren/Library/Arduino15/packages/digistump/hardware/avr/1.6.7/cores/tiny/Arduino.h:4,
                 from sketch/sketch_jul25a.ino.cpp:1:
/Users/warren/Library/Arduino15/packages/digistump/hardware/avr/1.6.7/cores/tiny/Print.h:75:18: error:   overriding 'virtual void Print::write(const uint8_t*, size_t)'
     virtual void write(const uint8_t *buffer, size_t size);
                  ^
exit status 1
Error compiling.
````

rmeister, 2016-07-25:
In the Arduino IDE, I had to choose the ATtiny85 in two steps: first the boad `ATtiny 25/45/85` and then the processor itself.

jywarren, 2016-07-25:
Hmm, I don't see those. I'm on Arduino IDE 1.6.7, will check for more recent version.

rmeister, 2016-07-25:
Wondering if there is a different between SoftSerial and SoftwareSerial? In the Arduino reference it is always called SoftwareSerial.

jywarren, 2016-07-25:
Yes, SoftSerial is a Digispark-provided library I've gotten to work on an ATTiny before, thought I'd try it as an alternative.

jywarren, 2016-07-25:
I'm now Arduino IDE 1.6.9, and I don't see `ATtiny 25/45/85` in the list; did you install extra boards? 

I have to run now, but can give this another go tomorrow. Thanks!

rmeister, 2016-07-25:
Yes I've added this URL to the boad manager: https://raw.githubusercontent.com/damellis/attiny/ide-1.6.x-boards-manager/package_damellis_attiny_index.json
And then installed the 'attiny' package:
[image]

jywarren, 2016-07-25:
Oh great, I'll try that.

rmeister, 2016-08-01:
Sorry, that package was the wrong track. 

From your error log I can see that you already installed the right one: the digistump package. Digistump brings its own implementation of the Print class, which is why the return types of `write()` are different. At the moment, I solved this problem with conditional compilation.

Also got a Digispark on friday, and to some extend it works with SoftModem (I managed to send data, but no receiving). Though, there is an issues with the ATtiny85: The analog comparator (used for receiving) shares pins with timer 0 (used for generating signals = sending) and therefore interfere, e.g. receiving while sending will not work.

An other problem could be the accurracy of the system clock, that is about 10% if not connected to an USB device. 
http://digistump.com/wiki/digispark/tricks#detect_if_system_clock_was_calibrated
Depending on the system's clock, the generated signal will vary.

jywarren, 2016-08-01:
Ah, interesting -- great sleuthing. So what are your next steps or needs?

rmeister, 2016-08-01:
Think I'll evaluate if it is possible to alternate sending and receiving. 

What I forgot to mention is the possibility to change one of the comparator input pins and use an ADC pin instead. But this will disable all other ADC pins. As the purpose of this project is communication for _sensors_, this is probably not beneficial.

rmeister, 2016-08-01:
And if you feel lucky, you can give your digispark a try :smile:  https://github.com/rmeister/SoftModem/tree/attiny-support
The output is on pin 1.

rmeister, 2016-08-03:
Leonardo will also be able to generate now. TX pin is 3, like for the Uno.

jywarren, 2016-08-04:
hoping to get to this tonight; we'll see!

jywarren, 2016-08-05:
I was able to successfully upload the test sketch onto both a Leonardo and a Digispark; I'll try attaching a headphone cable tomorrow!

jywarren, 2016-08-05:
The Leonardo uploaded with the simple example sketch you provided here: https://publiclab.org/notes/rmeister/07-18-2016/webjack-testers-needed#2.+Load+this+sketch+to+the+Arduino

jywarren, 2016-08-05:
Digispark blinks promisingly but I realized I have the wrong headphone plug... I'm going to try to grab another and modify but I am travelling this week, so hard to say.

rmeister, 2016-08-05:
Cool, blinking sounds promising. Thanks for your effort!

jywarren, 2016-08-05:
I wasn't able to find an affordable four pin cable at the airport, but I'm
going to try to get headphones on the plane and attach a speaker to the
digispark, then hold it up to the laptop microphone.

...maybe after I land.

Else I may be able to find a cheaper cable or a broken apple headset
soonish. I actually have both a Leonardo and a digispark with me. I have a
nanite too, but not sure I can flash that from codebender, and only have a
Chromebook at the moment.

Eventually I hope well be able to program over audio too :-)

jywarren, 2016-08-15:
OK, I was able to connect the Leonardo to the audio in, and confirm it's being played into the computer. This was with no extra circuitry -- just directly into the audio port. 

https://i.publiclab.org/system/images/photos/000/017/613/original/leonardo-webjack.mov

But I didn't see it read by the webjack demo. Maybe it's blowing out with too much amplitude?

rmeister, 2016-08-15:
[image]

Looks a little distorted. But one can see the preamble and some bits following. Maybe you can reduce the amplifying of the mic?

jywarren, 2016-11-27:
Sorry, moving last question to https://github.com/arms22/SoftModem/issues/5