publiclab/webjack #18: API method for get/set bitrate

jywarren, 2016-07-23:
and demo this by displaying bitrate in the demo, potentially allowing it to be changed

this may involve documenting the bitrate options that work on various platforms and in common

we also may not be ready to do this yet (fully), but can circle back and do this once it's more mature

Thanks!

rmeister, 2016-08-19:
Sorry for not answering to this for so long. 

For use with SoftModem, 1225 bit/s  work pretty well, even better than some slower ones. With SoftModem it also will not get faster than that, so the bitrate can be seen as a constant.

I think setting the bitrate is only necessary as soon as browser-to-browser transmissions via air do work. Because then we have higher frequencies that will eventually allow higher bitrates as well.

jywarren, 2016-09-06:
Complete in #42!

closed 2016-09-06 by jywarren