publiclab/webjack #20: Implement a RPC feature (like Firmata)

rmeister, 2016-07-25:
Less interesting due to overhead or dependency constraints:
https://github.com/wheeler-microfluidics/arduino-rpc
https://github.com/cloud-rocket/arduino-json-rpc
https://bitbucket.org/sources/tiny-rpc/wiki/Home

Probably the better choice:
https://github.com/thijse/Arduino-CmdMessenger
https://github.com/firmata/firmata.js

All of this RPC-like implementations currently require either UART or some kind of Wifi/Ethernet shield. For some, there are no JavaScript clients available or only node implementations.

To avoid inserting dependencies/requires in WebJack, I'd say writing a wrapper for use with Frimata and/or CmdMessenger is the better option.

@jywarren , what's your opinion?

jywarren, 2016-07-26:
awesome -- i agree and firmata.js looks like the best option.

rmeister, 2016-08-15:
https://github.com/publiclab/webjack-firmata

I'll close this one if you don't mind.

jywarren, 2016-08-15:
Great. Hi, Richard - I don't see any recent commits to webjack itself -- is the gh-pages branch showing the latest version?

closed 2016-08-15 by jywarren