publiclab/webjack #24: loopback in browser

mvayngrib, 2016-08-05:
(very cool library, looking forward to using it!)

i think the loopback is already there, it's processing audio from the microphone, it just doesn't decode it correctly yet :) I tried the master and fix-decoder branches. Does it work for you?

jywarren, 2016-08-05:
I haven't gotten it to loopback in the same browser or between mic and
speakers. Is there any filtering in the audio processing for suppressing
echo?

I'm interested in if it's possible to pass data between webpages this way.

rmeister, 2016-08-06:
I don't have my notebook with me, so I can't set up the gh-pages of my fork with the fix-decoder branch. Browser to browser transmission should do better with that fixes.

Seems like the web audio API has echo cancellation build in. Eventually it can be turned off: http://stackoverflow.com/questions/31306309/how-to-disable-web-audio-analyzer-filtering-high-frequencies

Interestingly I could observe loopback even with a cable attached, when using with SoftModem. There is crosstalk between the signal lines. Tried to get rid of it by sending the inverted signal on the other audio channel, but without success. I can post a picture of the crosstalk tomorrow. Any ideas on how to suppress this loopback are very welcome :-)

rmeister, 2016-08-08:
See #25 for a picture of the crosstalk.

rmeister, 2016-08-15:
Just tested loopback with echo cancellation turned off, but it did not make a big difference. 

When I placed the loudspeaker approx. 5cm away from the microphone and set the volume really loud, it did loopback to the same and to other tabs.  

However, with a cable connected (just one end, the other one is unconnected) the loopback is almost perfect :worried:

jywarren, 2016-08-15:
I'm guessing the latest gh-pages does not have the fix-decoder branch merged, so this may not be any new info, but I tried this out again with browser-to-browser and got some transmissions (holding speaker to mic) but pretty inconsistently (under 20% maybe). 

[image]

rmeister, 2016-08-15:
Correct, fix-decoder is not merged yet, but we can do that now.

jywarren, 2016-08-15:
OK -- go ahead, and I can use that in my tests (back in town now and have
access to things/tools/soldering). Thanks!

dwblair, 2016-09-23:
Posted same links to another issue, but in case it's interesting / useful -- here's an example of browser looping back into browser with another library / codebase:

https://cho45.stfuawsc.com/WebAudio-Modem/FSK/modem.html
https://github.com/cho45/WebAudio-Modem/tree/master/FSK