publiclab/webjack #25: Crosstalk

rmeister, 2016-08-08:
[image]

Picture above shows the microphone input. The received signal clearly contains a copy of the transmitted signal. Altough the crosstalk has a much lower amplitude than the proper signal, the decoder sometimes does recognize and decode it (but not always correctly).

Idea to get around this: send the opposite signal on the right audio channel to cancel out the crosstalk. Therefore, the right channel has to have exactly the same physical parameters as the left channel.

Coming back to this when I have the parts to build the second link.

jywarren, 2016-08-09:
Wow, interesting at the very least! Does this only happen when transmitting over open-air audio? I agree that it's a lower priority, but fun to think about. Perhaps we could try to detect this by listening for it in the blank spots, and "activate" compensation. Maybe I don't understand completely, but would it not be possible to send the cancelling signal on the same channel, additively? Again, just curious.

rmeister, 2016-08-09:
Oh I'm sorry, I did not describe the problem precisely. By crosstalk I mean capacitive or inductive coupling between two wires. The data sent on the left audio channel influences the voltage level on the microphone input line, because they run in parallel for a long distance. Like an open-air transmission this can cause loopback.

I don't really understand what you mean by listening in the black spots. Probably that was refered to open-air transmissions? For cable transmissions, we don't have to listen for crosstalk to activate compensation. We can send a compensating signal on the right audio channel right away when sending the actual signal.

Adding a cancelling signal to the left audio channel would cancel out everything == no transmission at all.

jywarren, 2016-08-09:
Perfect, that's helpful also -- in open air we might call it "feedback",
thanks for the clarification

rmeister, 2016-08-17:
The simplest solution to this is using a shorter cable, or at least a cable with more distance between the single wires.

rmeister, 2016-08-22:
Just a short note on the idea of sending the opposite signal at the second channel: tested this with just the cable attached (no Arduino) and it did not cancel out the crosstalk.