publiclab/webjack #26: API method for getting/setting frequencies

jywarren, 2016-08-15:
in looking at other applications for this library like browser-to-browser, i was thinking we could make this get/settable. 

https://github.com/quiet/quiet-js can operate at ~19kHz:

> For speaker transmission, there is a profile which transmits around the 19kHz range, which is essentially imperceptible to people (nearly ultrasonic).

rmeister, 2016-08-16:
Maybe it makes sense to introduce some kind of profiles as constructor parameter? There could be a profile for browser-to-browser communication at higher frequencies, and a profile for use with SoftModem that has echoCancellation enabled to encounter crosstalk.

jywarren, 2016-08-16:
Sure, well, we could read in an options object and we could provide a set
of common ones for different profiles, yes!

rmeister, 2016-08-24:
#29

jywarren, 2016-08-29:
And in #37, making an `connection.options` object that is a getter. This is probably sufficient for now, as we can re-initialize the whole thing to set -- but there is discussion in #37 on creating a `setOptions` method so setting can potentially be done after initialization.

jywarren, 2016-09-06:
Complete in #42!

closed 2016-09-06 by jywarren