publiclab/webjack #30: add volume "level" indicator to demo

jywarren, 2016-08-22:
to help in debugging -- it's good to see that, for example, something blinks when it hears *anything*. Maybe we can just include an existing module for this, and/or blink just one dot called "Volume" or "Sound detected"?

jywarren, 2016-08-22:
Could also link to https://test.webrtc.org/

rmeister, 2016-08-22:
That's an excellent idea, and maybe I can incorporate parts of it into a first-timer-only issue?

jywarren, 2016-08-22:
sure! Good idea, though maybe a `help-wanted` issue as a FTO issue should
be something you can spell out in a series of concrete steps. But if you
think it'll be that easy, go for it!

jywarren, 2016-08-29:
If you could point at a line in the decoder where there's a way to listen or query for activity -- it could just be a place to listen for a 1 or 0, instead of actually listening for volume -- that'd be really helpful -- could this line where `state` is changed to `DATA` work?

https://github.com/publiclab/webjack/blob/master/src/decoder.js#L178

rmeister, 2016-08-29:
Hm do we really want this stuff inside the code? 

Maybe we can use this example, but only in the code of the demo site:
https://webaudiodemos.appspot.com/volume-meter/index.html

jywarren, 2018-02-07:
Agreed, this should be in the demo code, and can be pretty much separate from core WebJack.

jywarren, 2018-02-07:
I like https://webaudiodemos.appspot.com/volume-meter/index.html !!!

jywarren, 2019-03-27:
Update!  I made a https://p5js.org sketch of an equalizer -- you can also hover your mouse to see the frequency:

https://editor.p5js.org/jywarren/sketches/TsICFM5ZO

It embeds, though maybe we'd need to run it from within the demo for cross- domain security reasons?

```html
<iframe src="https://editor.p5js.org/jywarren/embed/TsICFM5ZO" style="border:0;width:100%;height:400px;"></iframe>
```

Open in a new window here: https://editor.p5js.org/jywarren/full/TsICFM5ZO

jywarren, 2019-03-27:
[image]

jywarren, 2019-03-28:
I think i got this working!

[image]

jywarren, 2019-03-28:
I'll leave this open, because we could later expand the frequency range to include browser frequencies of 20,000Hz:

https://github.com/publiclab/webjack#profiles

I can't see how we can set the range the FFT p5JS plugin monitors using options here: 

https://p5js.org/reference/#/p5.FFT
https://p5js.org/reference/#/p5.FFT/analyze

But perhaps we could propose an upstream change to set the range of the FFT -- building on this issue:

https://github.com/processing/p5.js-sound/issues/353

jywarren, 2019-04-16:
Done!!!

closed 2019-04-16 by jywarren