publiclab/webjack #33: Diagram of wiring on Readme

jywarren, 2016-08-26:
perhaps a Fritzing diagram, as they're really readable?

jywarren, 2016-08-26:
I wasn't able to figure out what combination of wires went to what combination of Arduino pins; 
[image]

jywarren, 2016-08-26:
I know the capacitor/resistor circuit is important here, but even with this picture of that circuit I wasn't sure which headphone pins were connected to which Arduino Uno pins, and this looks like a one-way setup: https://publiclab.org/notes/rmeister/07-18-2016/webjack-testers-needed

rmeister, 2016-08-27:
I acutally tried to make a diagram in Fritzing, but I find it not very readable in this case. 

[image]

From the picture one can not see the labeling of the headphone jack pins. Here is the Fritzing file, where you can hover over the pins to see the labels:

[SoftModem.zip](https://github.com/publiclab/webjack/files/440666/SoftModem.zip)
(you have to rename it back to .fzz)
 
The circuit from the research note is indeed a one-way setup. A wanted to keep it simple, as we were only interested in testing the decoder of WebJack.

Although I would **_not recommend_** to do this, here is the most trivial bidirectional setup:
Connect the tip to pin 6, the second ring to a ground pin and the sleeve to pin 3.

jywarren, 2016-08-29:
I actually love this diagram -- I wonder if we could just:

a. run the leads off the edge of the board, and put a label (mic/left/right/ground) by each lead
b. show the corresponding pins on the most common 4-pin plug (male)

Something like this (with the plug standard marked also):

[image]

rmeister, 2016-08-29:
Sadly Fritzing has no labels for the breadboard view, so I used a note.
[image]

From http://allaboutwindowsphone.com/flow/item/17561_Jays_AB_headset_for_Windows_Ph.php

[image]

(pinout of your drawing is reversed)

jywarren, 2016-08-29:
> headset_for_Windows_Ph.php

yeah, sorry, i just made up the pins for my drawing, should've mentioned.
This looks good, and although I'm sorry to ask for one more tweak, I wonder
if we could:

* show the unused "right" zone between the mic and left labels, like right
(unused)

I was going to also suggest that we change the colors so that they
correspond to the typical AV cable colors -- right=red, left=white,
yellow=video -- but I realize that microphone != video, so this'd be pretty
confusing. Unless I'm mistaken. But if they are close-but-not-same, perhaps
we should actually not use /any/ of the same colors, to avoid confusion. We
could switch yellow for... purple? Green?

rmeister, 2016-08-29:
[image]

like this?

Colors are arbitrary by intention, but of course I can change yellow to something else ;)

jywarren, 2016-08-29:
Oh, sure they're arbitrary -- i just think that when we imagine someone
trying to build this, converting from Arduino pins to cable pins to
smartphone plug standard is hard enough, so I think we should try not to
confuse them by changing colors (or using colors which are *almost* the
same but not). They might get confused since the wires in the plug cable
itself might be color coded differently. Yes, that note looks great!

rmeister, 2016-08-29:
From what I have seen, there is no coloring standard for the wires inside the cable.

jywarren, 2016-08-29:
I've torn apart a few, and I've seen them match the yellow/red/white/black
to the RCA plug colors, which makes sense. I think it'll be fairly common
for people to cut open an RCA-style AV cable to get a 4-pin 3.5mm plug. But
I could be wrong about this being a common convention. In any case it's not
a big deal, the current color scheme is fine, just trying to think ahead to
common troubleshooting issues people may encounter.

rmeister, 2016-08-29:
Well, mine is green/black/red/white :sweat_smile:

jywarren, 2016-08-29:
haha -- well, just out of curiosity, was it an 3.5mm to RCA cable
originally? and i wonder if your green is my yellow? In any case i guess
you're right, we should not assume what's in a given cable.

rmeister, 2016-08-29:
3.5mm male to male, 4pin

jywarren, 2016-09-18:
Great -- one more thing to include -- the pinout for A/V 4-pole 3.5mm jacks -- but according to http://pinoutsguide.com, those pinouts are not very standardized: http://pinoutsguide.com/Home/av_jack_pinout.shtml -- but we could just link to that. 

| Pin # | Position | OMTP AV connector | AHJ AV connector |
|----|----|----|----|
| 1 | tip | Ground | Microphone |
| 2 | ring | Microphone | Ground |
| 3 | ring | Right audio | Right audio |
| 4 | sleeve | Left audio | Left audio |
| | **Devices:** | Samsung,older Sony Ericsson and Nokia | HTC, recent Sony and Nokia, Apple |

3.5mm to RCA AV cables (to red/white/yellow "TV-style" plugs) vary: [see this listing for various pinouts](http://pinoutsguide.com/Home/av_jack_pinout.shtml)

```markdown
| Pin # | Position | OMTP AV connector | AHJ AV connector |
| 1 | tip | Ground | Microphone |
| 2 | ring | Microphone | Ground |
| 3 | ring | Right audio | Right audio |
| 4 | sleeve | Left audio | Left audio |
| **Devices:** | Samsung,older Sony Ericsson and Nokia | HTC, recent Sony and Nokia, Apple |

3.5mm to RCA AV cables (to red/white/yellow "TV-style" plugs) vary: [see this listing for various pinouts](http://pinoutsguide.com/Home/av_jack_pinout.shtml)
```

closed 2016-09-20 by jywarren

selevo, 2021-07-23:
brouser to brouser send data for tes - not work ?
i click button on page on android phone №1 but log screen clear on android phone №2

jywarren, 2021-07-23:
Hi! Could you open this in a new issue, so we can work this problem separately from the one above? Thank you!

Are you hearing audible signal between the two? Are you able to transfer between a phone and a laptop browser? Sorry for the questions, just trying to narrow it down. If you can make a video of the process, that's even better.

jywarren, 2021-07-23:
You can also test it using this pre-recorded transmission to see if the issue is on send or receive: https://webjack.io/#plotting

commit eee9cb92 (2016-09-20): more documentation of wiring, close #33 (#48)