publiclab/webjack #43: iOS adapter

jywarren, 2016-09-02:
I remember @rmeister tried a hack to get iOS working and it didn't work -- can you paste in a link so perhaps someone could follow up and try to get it working?

rmeister, 2016-09-02:
http://iswebrtcreadyyet.com/#getUserMedia

It's hard to say if and when Apple will implement the getUserMedia interface, that we use here for WebJack.

Here's a link for a FSK generator script playing audio in Safari:
https://github.com/sixteenmillimeter/Javascript-FSK-Serial-Generator-for-Mobile-Safari

I gave it a short try, but I did not work out of the box.

jywarren, 2016-09-04:
What would be minimum specs necessary for an iOS adapter? Microphone access, of course. Just looking at this for generating audio in iOS: http://www.waudjs.com/api/

jywarren, 2018-02-10:
iOS now implements getUserMedia in iOS 11 and Safari 11!

jywarren, 2018-02-15:
It doesn't work yet and a number of WebRTC demos don't work with it yet. https://bloggeek.me/webrtc-ios-support/ has some details, but not much... i got an error with something like "audio devices found" when i tested on a borrowed iPhone. 

iPhones after v5 will support iOS 11 + Safari 11 which includes this.

https://github.com/addyosmani/getUserMedia.js/ is not super actively worked on and doesn't mention this in issues... 

Some discussion and an apparent working implementation over at https://github.com/processing/p5.js/issues/2146 -- cool!

jywarren, 2018-02-15:
aha! https://github.com/processing/p5.js/pull/2147/files

jywarren, 2018-02-28:
Good discussion on `adapter.js`  here for a cross-browser shim that allows iOS!

https://github.com/webrtc/adapter

https://github.com/webrtc/adapter/issues/685

They say:

> For me getUserMedia + adapterjs works fine on iOS11. The only thing I needed to add to video tag was:

```
  autoplay: 'autoplay',
  playsinline: 'playsinline',
```

Sounds positive!

jywarren, 2018-02-28:
https://webrtc.github.io/adapter/adapter-latest.js

jywarren, 2018-02-28:
If someone is able to add that link on this line:

https://github.com/publiclab/webjack/blob/5e8becee853fb2862be4cb1cc681c2fd2ffa3b57/examples/index.html#L9

and test it out on an iOS 11 Safari device, then we can delete this: https://github.com/publiclab/webjack/blob/master/examples/adapter.js

And add the above `autoplay` and `playsinline` params -- @rmeister these would normally be to the `<video>` element, but in this implementation do we create an `<audio>` element? 

```js
video.setAttribute('autoplay', 'autoplay');
video.setAttribute('playsinline', 'playsinline');
```

rmeister, 2018-03-01:
Yes there is no video element.
Eventually I can have a look at it in about one month when I return from travelling. Thanks for the research!

jywarren, 2018-03-01:
Thanks and no worries, just leaving an active trail of research here. Maybe
someone else will be able to solve it in the interim!

Safe travels!

jywarren, 2019-04-16:
Noting there is working iOS implementation in https://github.com/processing/p5.js/issues/2146, and we're also tracking at https://github.com/publiclab/spectral-workbench.js/issues/87 -- situation looks better, we probably just need to update our dependencies and tweak code as in the above p5js issue...

daz, 2019-04-17:
I got it going, couple minor changes https://github.com/publiclab/webjack/pull/91

jywarren, 2019-04-17:
oh wow!!! Thanks!!!!

jywarren, 2019-05-05:
Completed and confirmed!

closed 2019-05-05 by jywarren