publiclab/webjack #47: SoftModem profile works much better than Browser profile in open-air transmission

jywarren, 2016-09-16:
Please fill out the following information to help us address the bug you're filing.

### Hardware/Software

What board, circuit/shield, and smartphone/laptop are you using?

Firefox Android to Chrome/ChromeOS out loud (no cable) held up to microphone

As a test, I transmitted [the first paragraph of this article](http://www.newyorker.com/magazine/2014/04/14/deep-frieze) from my Android phone to the demo here: https://publiclab.github.io/webjack/examples/

I found that in the SoftModem profile (on both devices) it transmitted pretty well -- once the paragraph was done, almost all of it showed up:

[image]

But in the Browser profile, it did not transmit at all -- i got a strange y-shaped character, only, with multiple attempts. The sound was a lot quieter, but i held it right up against the mic, and that didn't make a difference.

The text is here: 

````
What will archeologists twenty-five centuries from now make of the ruins of One World Trade Center, currently nearing completion in downtown Manhattan? Some scholars in the year 4514, familiar with ancient accounts describing the diminutive structure as a “skyscraper,” will no doubt speculate about the significance of its height—although they will be forced to admit that 1,368, the number of “feet” from base to roof, was a figure with no known significance in the culture of the ancient builders. Others, drawing on fragmentary scriptural texts (“wikis”) that refer to a now missing aerial spire, will propose an ingenious theory that the original height of the building was 1,776 of those antique units of measure: a symbolic reference to a date known to have had considerable ideological importance for the builders. (Still others will dismiss this notion as vulgarly literal-minded.)
```

Maybe we should just stick with the basic SoftModem profile for the browser demo? Or was it for a more specific use case than I tried?

jywarren, 2016-09-16:
Actually, also, the quotation marks have been swapped for another character -- the text is flawless until the open parenthesis at the end, with the exception of those quote marks!

rmeister, 2016-09-17:
To be honest, the frequencies in the browser profile are almost untested. I just took the highest possible frequencies for a sample rate of 44.1kHz ( → max is 22.05kHz due to Nyquist) that are a multiple of the baudrate.

Quotation marks are eventually a matter of the used character set.  I don't think it is a transmission error, as there is a test case for all 256 possible bytes.

jywarren, 2016-09-17:
Ah, cool -- sometime it'd be great to make a demo which sets a new custom
profile from text inputs, and we (or others) could try to find
bitrate/frequency combinations that work best for a given use case.