publiclab/webjack #59: Trouble with while(modem.available()) when sending messages to Arduino

jywarren, 2016-11-22:
Please fill out the following information to help us address the bug you're filing.

### Hardware

What board, circuit/shield, and smartphone/laptop are you using?

Using a SoftModem shield (pins 3 and 6, standard) plus an iMac with a 4=>3 pin splitter. 

### Software

* What Arduino sketch, web browser, and WebJack profile are you using?

Default https://publiclab.github.io/webjack/examples/

* Attach a recording of the received signal. Please use this AudioRecorder: https://webaudiodemos.appspot.com/AudioRecorder/index.html

For debugging purposes, i also used this recording: https://www.youtube.com/watch?v=GtJW1Dlt3cg

* Include any error messages that occurred

I'm just not every receiving anything on the SerialBridge example: https://www.youtube.com/watch?v=GtJW1Dlt3cg

It also doesn't echo out "Listening" when i insert this within the `while(modem.available())` loop:

```c
    Serial.print("Listening");
```

That loop is here: https://github.com/publiclab/webjack/blob/master/sketches/SerialBridge/SerialBridge.ino#L14

I'm not sure why `modem.available()` isn't triggering... any ideas? 

For bug reports, fill out the above template; for feature requests, you can delete the template.

To learn how to write really great issues, which increases the chances they'll be resolved, see:

https://publiclab.org/wiki/developers#Contributing+for+non-coders

jywarren, 2016-11-22:
OK - I realized my splitter had microphone and speaker swapped. I can now send from the Arduino to the browser using this program:

https://gist.github.com/jywarren/09d1f93caf7f5e5e4a7dad2cf724bcad

However, I can't seem to send from the browser to the Arduino using that script. Any troubleshooting tips I should watch for?

rmeister, 2016-11-22:
Is it an Arduino Uno?

jywarren, 2016-11-22:
Yes, apologies for forgetting to put that in.

rmeister, 2016-11-22:
I guess you already tried different volume levels?

You could activate debug mode for SoftModem and look if the led blinks when you play/send something from the browser:

    #define SOFT_MODEM_DEBUG_ENABLE  (1)

jywarren, 2016-11-26:
OK, i'll try to do that, but don't have a computer capable of Arduino IDE at the moment (not at home). I also was able to get Arduino => Nexus 5 (Android) working and Arduino => Chromebook working, but not the reverse for either device. 

I'm curious if volume is an issue, because I can hear the Arduino's output (clearly but not loudly) with headphones plugged into the Arduino, and the laptop and phones' output volume is quite a bit louder than that. But I guess the two audio input interfaces are quite different, so who knows.

jywarren, 2016-11-26:
And just to be extra sure, you've def. gotten send/receive working using exactly the `/sketches/SerialBridge/SerialBridge.ino` sketch? 

Thanks, Richard!

rmeister, 2016-11-28:
Sorry @jywarren , overlooked your last question. Yes I did. 

I see you worked a lot on Webjack last days. I'll take a look at the changes on Thursday or Friday!

jywarren, 2016-11-28:
Could be related to https://github.com/publiclab/webjack/issues/61 -- needs more rigorous testing. Thanks!