publiclab/webjack #61: much higher success rate on USB power vs. 'noisy' AC adapter

jywarren, 2016-11-28:
Notes from my testing last night showed that a "wall wart" AC adapter created an audibly noisy signal from the Arduino, and a much lower success rate for 2-way communication. 

### Hardware

What board, circuit/shield, and smartphone/laptop are you using?

Uno, SoftModem prebuilt shield, iMac with TRRS cable splitter.

The recorded sound is here: https://publiclab.org/system/images/photos/000/018/895/original/myRecording00.wav

### Software

* What Arduino sketch, web browser, and WebJack profile are you using?

Using [webjack-firmata](https://github.com/publiclab/webjack-firmata/) and this is important because some applications need the extra power an AC adapter offers, like stepper motor drivers: https://github.com/publiclab/webjack-firmata/issues/21

rmeister, 2016-12-03:
[image]

Not sure if this is random noise. I expected to see some 50Hz ripples from the AC adapter, but it doesn't look like that. I'll watch out for an AC adapter to see if I can reproduce the behavior.

jywarren, 2016-12-03:
Hmm, interesting. Does anything here stand out to you as unusual? It sounded kind of harsher and clipped, compared to the usual, in my headphones. 

This kind of debugging reminds me of Forrest Mims Engineer's Mini Notebook: Science Projects:

[image]