publiclab/webjack #65: Differences in 49 baud transmission from Arduino, Android,

jywarren, 2016-12-27:
I created a new example profile which is identical to the standard SoftModem profile, but with 49 baud, which is a factor of 44,100 -- 44.1kHz -- to see if I could get better transmission at lower speeds, and as an explanatory example slow enough for the human ear to hear the high/low switches. 

https://github.com/publiclab/webjack/blob/master/examples/index.html#L113-L120

I then modified SoftModem.h to set the Arduino's transmission baud to 49 bits/second.

But playing each, I hear distinct differences between them -- do I need to adjust something else to get this to work? I wasn't able to send/receive on either a Mac laptop, Arduino (both with Chrome), or to transmit to either from an Arduino with these settings. 

Here are recordings of each:

https://publiclab.org/system/images/photos/000/019/144/original/arduino-49.wav

https://publiclab.org/system/images/photos/000/019/145/original/android-chrome-49.wav

https://publiclab.org/system/images/photos/000/019/146/original/mac-chrome-49.wav

Some don't even sound to me like the same pitch. Any advice -- and any additional guidance on choosing baud rates? I'd like to add to the documentation for this. Thank you!

rmeister, 2017-01-07:
49 Baud will not work, because the TCNT_BIT_PERIOD constant exceeds the range of an uint8_t and is therefore trunkated. As countermeasure we could try to change the prescaler, but then the error rates seem to get very high.

I've added some comments to this sheet that can be used to play around with the different parameters:
https://docs.google.com/spreadsheets/d/1eGkrUqYAaXadsdk2csurmt97gBlT8dznx2IJKigZT90/edit?usp=sharing

jywarren, 2017-01-09:
> dznx2IJKigZT90/edit?usp=sharing

Oh, excellent -- is there any way to export that so we can add it to the
repository?

Can you explain or link to more info on: relative error HIGH/relative error
LOW ?

Thanks a lot, Richard!

jywarren, 2017-01-09:
> mt97gBlT8dznx2IJKigZT90/edit?usp=sharing

Also, did you enter 126, 315, 630, 1225 as examples of baud rates that
would work?

rmeister, 2017-01-10:
>Oh, excellent -- is there any way to export that so we can add it to the
repository?

Maybe as LibreOffice file? 

>Can you explain or link to more info on: relative error HIGH/relative error
LOW ?

SoftModem uses integer arithmetics and therefore loses precision in signal parameter calculation. I looked at the difference between generated and ideal bit length. It is important to keep the difference small, elsewise the decoder gets out of sync.

>Also, did you enter 126, 315, 630, 1225 as examples of baud rates that
would work?

I added them as they are the proposed ones in SoftModem.

jywarren, 2017-01-10:
Awesome, thank you. I think the table could for starters be embedded in the
README statically, but yes, i think a LibreOffice file could also be
included in the repository?

Thank you, Richard!