publiclab/webjack #93: Equalizer/fft graph not shown on mobile phones

jywarren, 2019-04-16:
https://github.com/publiclab/webjack/issues/88, #87  and #92 fixed the audio security limits issue and installed an equalizer to display what sounds it can hear on the spectrum, for debugging. It's awesome! 

[image]

However, on my Android phone in Chrome 73, I can't see the graph:

(adding soon)

I can't reproduce this on desktop Chrome:

[image]

Not sure what's happening here!

jywarren, 2019-04-16:
Here's the section that's supposed to draw the graph!

https://github.com/publiclab/webjack/blob/04c0def1f5f489c65558154f548f707bdca85816/examples/fft/sketch.js#L40-L46

selevo, 2021-07-19:
test soul brauser on android.
Chrome browser hasn't been very good lately

jywarren, 2021-08-12:
aww darn, any way you could test with a javascript console? Thanks for
reporting!

selevo, 2021-08-14:
> aww darn, any way you could test with a javascript console? Thanks for reporting!
> […](#)

https://editor.p5js.org/jywarren/sketches/rkztwSt8M

#ReferenceError: webkitAudioContext is not defined
    at Class.init (https://webjack.io/dist/webjack.js:663:44)

    at new Class (https://webjack.io/dist/webjack.js:63:19)

    at (anonymous function) (https://jywarren.github.io/p5-webjack/lib.js:15:20)

    at (anonymous function) (https://cdnjs.cloudflare.com/ajax/libs/p5.js/0.8.0/addons/p5.sound.min.js:24:4838)(https://github.com/publiclab/webjack/issues/93#issuecomment-882318312)>, or unsubscribe <https://github.com/

jywarren, 2021-08-14:
Ah. Hm. I wonder if this is fixed in a more recent version of p5.js; we
seem to be at v0.8.0. I searched for the error in their repo but found
nothing but their current version is v1.4.0:
https://github.com/processing/p5.js/

We could try bumping our version...

jywarren, 2021-08-14:
Wow, we're actually at v0.7.3 of p5js:

https://github.com/publiclab/webjack/blob/5995944d8e353e01a555976ebeeb0e5a6a6b8460/examples/index.html#L16-L18

Would you be able to try bumping the version in the CDN links and trying again? If not I can try.

Noting also that the code referenced in your error log mentions this wrapper library I also wrote, but there's no reference to "webkit" in the code there so I think that's not the issue. But still, good to notice the included code:

https://github.com/jywarren/p5-webjack/blob/main/lib.js

jywarren, 2021-08-14:
thanks so much for helping out here!!!

selevo, 2021-08-15:
i dont know how get console js errors on android brausers whithout PC 
In this brausers not console

selevo, 2021-08-22:
> thanks so much for helping out here!!!
[webjack.io-1629558757386.log](https://github.com/publiclab/webjack/files/7027624/webjack.io-1629558757386.log)

[image]

jywarren, 2021-08-22:
oh whoa... cert invalid! This i think i can fix. I don't know if it'll fix
things on Android but it's a good first step to get out of the way.

jywarren, 2021-08-22:
That's odd... I see this:

[image]

Let me see what's up.

jywarren, 2021-08-22:
What version of Chrome are you using?

jywarren, 2021-08-22:
I don't know if this is the problem on Android but it makes me wonder, what is your Android version and Android Chrome version too?

chrome://settings/help can help you determine this. 

Would you be able to try out the other change as well, the update to p5js v1.4.0, to see if that helps? 

https://github.com/publiclab/webjack/pull/202 I've tried the update here. Do you know how to start it up to test? You could use GitPod -- 

http://gitpod.io/#https://github.com/publiclab/webjack/pull/202

once it starts up, you'll see the example page in the preview window.

selevo, 2021-09-06:
The interface there is very slow and it is intended for a computer, the page does not move well on the phone and the navigation is bad I can not check the scripts

selevo, 2021-11-19:
https://github.com/publiclab/webjack/issues/208