publiclab/webjack #103: CSV download of log

jywarren, 2019-04-23:
It'd be great to be able to download the log as a CSV!

jywarren, 2019-05-05:
```js
var csv = connection.history.received.join('\n'); // the CSV data

// old school, but doesn't seem to work for me?

window.open("data:text/csv;charset=utf-8," + escape(csv));
```

or generate a link: https://stackoverflow.com/questions/17564103/using-javascript-to-download-file-as-a-csv-file

```js
<a href="data:text/csv;charset=utf-8,'+escape(csv)+'" download="webjack.csv">download</a>
```

We should put the timestamp (and ideally the better-formatted date + time) in the filename; `new Date().toJSON().slice(0,10).replace(/-/g,'/');` gets the date, and we can add timestamp after that maybe...

jywarren, 2019-05-05:
Let's put the button here! 
https://github.com/publiclab/webjack/blob/ce5d2a943bd18e55911a7ecbef7702570e533164/examples/index.html#L55-L58

DevanshCodes, 2019-12-29:
Hey, if its still open, can I claim it?

jywarren, 2019-12-29:
Yes please! Thanks for the help!

sifihog, 2020-01-31:
Hello!
Just wondering if there's been any progress on this? I'd like to submit a pull request if its still open.