publiclab/webjack #119: Weekly Community Check In 35: Hacktoberfest is here

cesswairimu, 2019-09-30:
Please fill out the following information to help us address the bug you're filing.

### Hardware

What board, circuit/shield, and smartphone/laptop are you using?

### Software

* What Arduino sketch, web browser, and WebJack profile are you using?
* Attach a recording of the received signal. Please use this AudioRecorder: https://webaudiodemos.appspot.com/AudioRecorder/index.html
* Include any error messages that occurred

For bug reports, fill out the above template; for feature requests, you can delete the template.

To learn how to write really great issues, which increases the chances they'll be resolved, see:

https://publiclab.org/wiki/developers#Contributing+for+non-coders

cesswairimu, 2019-09-30:
@gauravano
@milaaraujo @kevinzluo @Rishabh570 @sashadev-sky @shubhscoder @sidntrivedi012 @SidharthBansal @stefannibrasil @ViditChitkara @tech4GT @mridulnagpal @IshaGupta18 @CoderJolly @namangupta01 @sagarpreet-chadha @Souravirus @MayankKashyap @siaw23 @ryzokuken @icarito @steviepubliclab, @ebarry, @jywarren, @sagarpreet-chadha, @JonathanXu1 @rexagod, @divyabaid16, @dinaelhanan, @oorjitchowdhary @publiclab/mentors @publiclab/image-sequencer-guides @publiclab/leaflet-environmental-layers-guides @Paarmita @publiclab/reviewers @Divy123 @bhavayAnand9 @Mridul97 @avsingh999 @Dhiraj240 @codeIriss @IgorWilbert @pdurbin @HarshKhandeparkar @niravasher @publiclab/reviewers @publiclab/is-reviewers @publiclab/community-reps @divyabaid16 @publiclab/community-reps @subhahu123 @govindjeevan @gautamig54 @rarrunategu1 @CleverFool77

cesswairimu, 2019-09-30:
@ananyaarun @ananya @publiclab-mimi @ananya @hc-barker @starkblaze01 @asquare @pdurbin @HarshKhandeparkar
@newbazz @aSquare14 @vaarigupta @alonpeer @aashna27 @harshithpabbati @jesus-montano @cryptoclidus @mohamed-elalem @konolga @gowtham5699
@jywarren @ebarry @steviepubliclab @StlMaris123

pydevsg, 2019-10-01:
I am super excited to be part of Google Code In contest and would love to mentor this year 😄

jywarren, 2019-10-01:
:-) awesome! Thanks all! Welcome to Hacktoberfesters!

Also noting that the Outreachy winter session contribution period opens today! Welcome to Outreachy applicants!

https://github.com/publiclab/plots2/issues/6285

All this means we should work to convert some `fto-candidate` issues into full-fledged `first-timers-only` issues!

https://github.com/publiclab/plots2/labels/fto-candidate

https://github.com/publiclab/plots2/labels/first-timers-only

Thanks, everyone!!!

pydevsg, 2019-10-01:
I will surely create more __first-timers-issue__ and will review them as much as possible , helping them contribute to the PL community 😄 .

ebarry, 2019-10-02:
Hello everyone!
I would like your advice on possibly highlighting these Education-related and moderation-related issues for consideration in Hacktober? Do these look ready to go? Are there any questions right off the bat i could answer? Let me know!
Thank you!

Issues related to Education:
https://github.com/publiclab/plots2/issues/6281 - Moderation: move spam off dashboard
https://github.com/publiclab/plots2/issues/6282 - Moderation: change time of mod emails
https://github.com/publiclab/plots2/issues/6244 - Moderation related, but not related to education:
https://github.com/publiclab/plots2/issues/6173 - Ed-related: activity:foo should take wiki page “foo”

SidharthBansal, 2019-10-07:
Oh great @cesswairimu 
I have labeled some issues with hack October so that Public Lab repository is visible on their website. 
I have also opened up an issue so that people can come at PL too.

SidharthBansal, 2019-10-07:
Here is the link https://github.com/publiclab/plots2/issues/6412

cesswairimu, 2019-10-07:
awesome thanks @SidharthBansal

cesswairimu, 2019-10-15:
Moved to https://github.com/publiclab/plots2/issues/6470. Thanks all

closed 2019-10-15 by cesswairimu